autowarefoundation/autoware_universe #10051: Backward trajectory is generated when the ego vehicle invades a drivable area

Kim-mins, 2025-01-30:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

Hi team, 

I'm currently running Autoware with Carla, and I found a situation that a backward trajectory is generated when the ego vehicle invades a drivable area.

Here's the video of the situation: [[rviz](https://salbox.synology.me/drive/d/s/11xNTtzZfvKBendreTpLPoC7sLBMsW23/hQ84HhJYZQqpsI2WVkFpMpt_1apYxNwr-pLrgQl3zAww)]
In the video, the ego vehicle initially stepping a drivable area, and the generated trajectory is for driving backward.
At 0:29, the NPC vehicle drives to behind the ego and  the ego starts to turn left (I cannot sure the NPC has an impact on this issue), which makes the ego vehicle invades the center line.

I tried to reproduce the situation with [the planning simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/), and **I think the issue occurs when the ego vehicle invades the drivable area at the initial point**.
Here's the video for more details: [[planning simulation](https://salbox.synology.me/drive/d/s/11xNi3JQkWFFKZpA5aIke8BbQRYF0tqV/2c5QCtpALyPkPQreMNpKIl-Wb6-pa-7x-1rsg0iX0Aww)]
The video shows two cases:
1. (The ego does not invade the drivable area) At 0:06, the ego vehicle is initialized and drives to the destination well.
2. (The ego invades the drivable area): However, At 0:29, Autoware generates backward trajectory, and the ego vehicle reaches the destination only after following the trajectory.

Actually, I cannot point out the reason why the ego vehicle drives forward in the first video, even though backward trajectory is generated. (I don't know why the gear does not changed)
But I thought that the situation in the planning simulation is similar to that of the first video.

Here are more details of the situation:
- Starting/Goal pose (from `launch.log` file)
```
initial pose - x: 0.726764, y: 190.627121, z: 0.068133
initial orientation - qx: -0.000201, qy: 0.000178, qz: -0.715753, qw: 0.698354
goal pose - x: 209.993347, y: -9.808371, z: 0.275307
goal orientation - qx: 0.000000, qy: 0.000000, qz: -0.007468, qw: 0.999972
```
- Map: [Town03](https://drive.google.com/drive/folders/1rt1pKEfY-hqC2pMrHfajg1EVfjlIcAX5)
- ros2bag file: [[ros2bag](https://salbox.synology.me/drive/d/s/11xNWerWmWIQ2hXT5z8eFnhGgcKHRsZL/prM_15FK_y5RYxHkrAuGbt-XpjlnpMCV-Cr9AX0X2Aww)]
- launch.log file: [[launch.log](https://salbox.synology.me/drive/d/s/11xNgdFo2fvNwyv6lqilLQflkIUfT42d/e-pXWN0A2tFt32fa8Y6UAfCukH5TFJnK-H79AHlP2Aww)]

### Expected behavior

I hope the ego vehicle drives following the center of the lane.

### Actual behavior

However, the ego vehicle invades the center line.

### Steps to reproduce

1. Pull image from `https://github.com/autowarefoundation/autoware/pkgs/container/autoware/334254906?tag=universe-devel-cuda-0.40.0-amd64`, and run Autoware.
2. Run ros2bag file above.

In the planning simulation, I could utilize the initial/goal pose above.

### Versions

- OS: Ubuntu 22.04 (Docker), Ubuntu 20.04 (host)
- Autoware: [0.40.0](https://github.com/autowarefoundation/autoware/tree/0.40.0)
- Carla: 0.9.15
- autoware_carla_interface: [link](https://github.com/autowarefoundation/autoware.universe/tree/0.40.0/simulator/autoware_carla_interface)

Kim-mins, 2025-02-27:
Thank you for the response @kyoichi-sugahara!

I visualized the `start_planner` while replaying the [ros2bag](https://salbox.synology.me/drive/d/s/11xNWerWmWIQ2hXT5z8eFnhGgcKHRsZL/prM_15FK_y5RYxHkrAuGbt-XpjlnpMCV-Cr9AX0X2Aww) file and it seems the generated path is for drive backward and straight, and the shape is weird.
Here's a video in rviz: [[rviz](https://salbox.synology.me/drive/d/f/12KmQKMmjIrEa8VhOfw53Iw72Y7eRrm1)]

In the video, it seems there are two kinds of paths: one for backward driving (a blue one) and the other for forward (a green one).
However, the shape of the backward trajectory seems weird (ends with a sharp tip, and the shape is different with the green one).
Also, there's a blue line that starts from the baselink of the vehicle and it starts to deviate when the vehicle starts to move [[video](https://salbox.synology.me/drive/d/s/12Kmvuv12mX2EszsED6AjbWa25XjtdOZ/Bs_k6uxBzVw4MNQiL0ZaDP_pIW6XdlnR-LLwAeLsaFgw)].

Please let me know if you need more information. Thanks!

kyoichi-sugahara, 2025-03-10:
@Kim-mins 
I can't access to the link you shared 😢 
My colleagues also can't access...
I will try later but can you check whether this link is still valid or not?

Kim-mins, 2025-03-14:
Hi @kyoichi-sugahara!

I just recovered the link and I could access to the video. Could you please check if you have time?

Thanks!

kyoichi-sugahara, 2025-03-19:
@Kim-mins 
Can you visualize debug info which is added in this [PR](https://github.com/autowarefoundation/autoware_universe/pull/10029)?

I'm still suspicious about following point
> it appears that the bounding box generated from ego vehicle's current position is deviating from the current lane.