autowarefoundation/autoware_universe #10068: Create meta package that all CUDA-dependant packages will depend on

esteve, 2025-02-04:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

Following the discussion in https://github.com/autowarefoundation/autoware.universe/issues/3135, we'll have a meta package that all the CUDA-dependant packages will depend on so that we can filter them out and be able to build only the Autoware packages that don't depend on CUDA directly or indirectly.

### Purpose

Be able to filter out CUDA-dependant packages from Autoware.

### Possible approaches

Create a meta package that all CUDA-dependant packages depend on.

### Definition of done

A subset of Autoware that only has packages that do not depend on CUDA can be built.

xmfcx, 2025-02-04:
- I've also created https://github.com/autowarefoundation/autoware.universe/issues/10069

esteve, 2025-02-05:
@xmfcx closing this one since the one you created is more complete. Thanks

closed 2025-02-05 by esteve