autowarefoundation/autoware_universe #10069: Create a CUDA dependency meta-package

xmfcx, 2025-02-04:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

**Continuation from:** 
- https://github.com/autowarefoundation/autoware.universe/issues/3135#issuecomment-2634342143

### Purpose

To simplify filtering out packages that depend on CUDA.

### Possible approaches

Create a meta package with a basic CMakeLists that checks for CUDA.

All packages that **directly** depend on CUDA should depend on this meta package.

**Package name:** `autoware_cuda_dependency_meta`

The README of this package should explain why the package exists and should contain possible usages.

**Example:**
- `colcon build --packages-ignore $(colcon list --packages-above autoware_tensorrt_common autoware_cuda_utils --names-only | tr '\n' ' ')`

### Useful graph

`colcon graph --packages-above autoware_cuda_utils autoware_tensorrt_common --dot | dot -Tsvg -Grankdir=LR -Gnodesep=0.2 -Granksep=0.3 -Gconcentrate=true -o graph.svg`

[image]

### Definition of done

- [ ] Create the meta-package
- [ ] Update the package.xml files of the packages that directly depend on CUDA.

esteve, 2025-03-03:
The buildfarm builds fine when excluding the `autoware_cuda_dependency_meta` https://github.com/autowarefoundation/autoware-deb-packages/pull/127

esteve, 2025-03-10:
Closing this now that https://github.com/autowarefoundation/autoware.universe/pull/10073 has been merged and that https://github.com/autowarefoundation/autoware-deb-packages/pull/127 builds fine.

closed 2025-03-10 by esteve