autowarefoundation/autoware_universe #10150: CI: Ccache is too small

mojomex, 2025-02-18:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

When looking at [a random CI run](https://github.com/autowarefoundation/autoware.universe/actions/runs/13368521713/job/37335100701) I noticed that the Ccache size is limited to a very small `600 MB` whereas GitHub allows for payloads of up to [`10 GB`](https://docs.github.com/en/actions/writing-workflows/choosing-what-your-workflow-does/caching-dependencies-to-speed-up-workflows#usage-limits-and-eviction-policy) in their cache action.

The observed cache hit ratio is quite poor as a result:
```
525M	/root/.ccache
Summary:
  Hits:             2867 /  7670 (37.38 %)
    Direct:         1671 /  7892 (21.17 %)
    Preprocessed:   1196 /  6004 (19.92 %)
  Misses:           4803
    Direct:         6221
    Preprocessed:   4808
  Uncacheable:      5398
Primary storage:
  Hits:             5667 / 16696 (33.94 %)
  Misses:          11029
  Cache size (GB):  0.55 /  0.60 (91.25 %)
  Cleanups:           34
```

### Expected behavior

I'd expect the Ccache to be limited to 10 GB, and to have hit rates of close to 100% for PRs such as the one that triggered the above CI run: https://github.com/autowarefoundation/autoware.universe/pull/10116

### Actual behavior

The cache has only 600 MB and a hit rate of under 40% for a PR with no code or build system changes.

### Steps to reproduce

See the above CI pipeline run

xmfcx, 2025-02-18:
It is limited to 600MB because Github Public Runners don't have much space available.

Please see:
- https://github.com/autowarefoundation/autoware.universe/pull/7555

Maybe with current `ubuntu-24.04` images, we might have slightly more space but it's a tight rope.

mojomex, 2025-02-18:
Thanks for the fast respose 🚀 
I see...  sad times 😥

closed 2025-02-18 by mojomex

mojomex, 2025-08-19:
As posted [here](https://github.com/autowarefoundation/autoware_universe/issues/10932#issuecomment-3199109952):

We could also introduce a [Ccache Remote Storage Backend](https://ccache.dev/manual/4.7.html#_remote_storage_backends).

A full Autoware build produces > 15 GB of Ccache, but as seen in the comments above, this cannot be fixed since our runners have limited hard drive space (#7555).

A remote cache can be shared across all runners, workflows and steps, and will handle changed dependencies, different compilers etc. correctly. It would however require a hosted Redis instance. Cost-wise this should be quite cheap, and might be offset by saving AWS costs when building CUDA code that would also utilize this new backend.

esteve, 2025-08-19:
The disk space limit was back when we were using public runners, right @xmfcx ? Don't we use hosted runners nowadays?

xmfcx, 2025-08-19:
https://github.com/autowarefoundation/autoware_universe/blob/8c162649aa7e7319b8c8c01c3896500f0b4e5a51/.github/workflows/build-test-tidy-pr.yaml#L43-L76

| Workflow                                   | Runner Type     |
|--------------------------------------------|-----------------|
| build-and-test-differential                | Public runners  |
| build-and-test-differential-cuda           | Self-hosted     |
| build-and-test-packages-above-differential | Self-hosted     |

And here is the check for cuda-necessity:

https://github.com/autowarefoundation/autoware_universe/blob/8c162649aa7e7319b8c8c01c3896500f0b4e5a51/.github/workflows/build-test-tidy-pr.yaml#L29-L36

And even if cuda is required for a PR, both `build-and-test-differential` and `build-and-test-differential-cuda` are run in parallel to make sure the build is still successful on non-cuda version too.

So, many (more than half) of the CI runs are made in public runners.

xmfcx, 2025-08-19:
Will ~~sacrifice~~ re-allocate one of the 8 self hosted runners to host a Redis cache with 24GB of effective cache with a single Redis instance. The machine is 8c16t and 32GB RAM, since Redis is single-threaded it will be slightly underutilized but it's a good way to start testing.

This is not a very urgent issue for now, since [here](https://github.com/autowarefoundation/autoware_universe/actions/runs/16671491300/job/47188653507) the modification of `pointcloud_preprocessor` on `build-and-test-differential-cuda` took 6 mins to compile, I think it's reasonable.

But definitely worth giving it a shot. If it works, it would simplify the CI since we wouldn't need to maintain a ccache cache with github, instead would pass a password to the redis and they all would pull/push from the remote cache.

esteve, 2025-08-19:
@xmfcx 

> I'll also test how https://ccache.dev/manual/4.7.html#_remote_storage_backends could work on my machine.

Please see question above about whether we need this with out self-hosted runners, I'd prefer using something simpler than more infrastructure (e.g. Redis). See for example the following GitHub action:

https://github.com/marketplace/actions/local-cache-on-self-hosted-runner

esteve, 2025-08-19:
> > https://github.com/marketplace/actions/local-cache-on-self-hosted-runner
> 
> This probably wouldn't work because with build-and-test-differential workflow, we don't operate on the runner directly, we utilize [container:](https://github.com/autowarefoundation/autoware_universe/blob/731e345663d726210a63077c072ab5bf4af5ca57/.github/workflows/build-and-test-differential.yaml#L43) argument, so it is isolated from the system.

The host can share a volume with the container. It might even have better performance, given that we'd be accessing the disk directly:

https://docs.github.com/en/actions/reference/workflows-and-actions/workflow-syntax#jobsjob_idcontainervolumes

But it'd only work for self-hosted runners, of course, not public ones.

xmfcx, 2025-11-17:
I've made some tests today regarding the ccache on: large vs small cache and high vs low compression.

# Tests

## Test description

**Test 1: Large cache**
1. Clear & reset cache, limit 500GB
2. Remove build,install
3. **First run:** Build & log build time & cache utilization
4. Remove build,install
5. **Second run:** Build & log build time & cache utilization

**Test 2: Small cache**
1. Clear & reset cache, limit 600MB
2. **First run:** Remove build,install
3. Build & log build time & cache utilization
4. Remove build,install
5. **Second run:** Build & log build time & cache utilization

**Test 3: Large cache, compress**
1. Clear & reset cache, limit 600GB, compression level = 3 (default is 1)
2. **First run:** Remove build,install
3. Build & log build time & cache utilization
4. Check the cache size for different compression levels

## Machine specs & test env

- 5900X (12c 24t)
- 109GB DDR4 RAM

Performed on my desktop, closed all other applications but still expect very low outside utilization.
Each build is performed on a new clean terminal to prevent env var creep.

```bash
cd autoware
git pull
git checkout main
vcs import src < autoware.repos
vcs import src < autoware-nightly.repos
vcs import src < tools-nightly.repos
colcon build \
  --event-handlers console_cohesion+ \
  --cmake-args -DCMAKE_BUILD_TYPE=Release \
  --mixin coverage-gcc coverage-pytest compile-commands
```

Mostly relevant env vars in `.bashrc`:
```bash
export CMAKE_C_COMPILER_LAUNCHER=ccache
export CMAKE_CXX_COMPILER_LAUNCHER=ccache
export CCACHE_DIR="$HOME/.cache/ccache/"
export CCACHE_LOGFILE=/tmp/ccache.log
export CMAKE_C_FLAGS='-fdiagnostics-color'
export CMAKE_CXX_FLAGS='-fdiagnostics-color'
```

```bash
$ cat ~/.cache/ccache/ccache.conf
# Set maximum cache size
max_size = 600M  # Modified for both tests.
```

## Test results

### Test 1: Large cache

#### First run

- Summary: 497 packages finished [42min 25s]
[... 180 lines cut ...]
New compressed data:   733.2 MB (10.0% of original size)
  - Compression ratio: 9.997 x  (90.0% space savings)
Size change:           -72.9 MB
mfc@whale:~$ ccache --recompress 19
Recompressing... 100.0% [==============================================================================================]

Original data:           7.3 GB
Old compressed data:   733.2 MB (10.0% of original size)
  - Compression ratio: 9.997 x  (90.0% space savings)
New compressed data:   660.2 MB (9.0% of original size)
  - Compression ratio: 11.102 x  (91.0% space savings)
Size change:           -73.0 MB
mfc@whale:~$ ccache --recompress 1
Recompressing... 100.0% [==============================================================================================]

Original data:           7.3 GB
Old compressed data:   660.2 MB (9.0% of original size)
  - Compression ratio: 11.102 x  (91.0% space savings)
New compressed data:   896.6 MB (12.2% of original size)
  - Compression ratio: 8.175 x  (87.8% space savings)
Size change:          +236.4 MB
```
</details>

[image]