autowarefoundation/autoware_universe #10159: clang-tidy-differential workflow doesn't work

mebasoglu, 2025-02-18:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

The `build_depends.repos` file was separated into two in the following PR:
- https://github.com/autowarefoundation/autoware.universe/pull/10148

However, the clang-tidy-differential workflow still uses the old `build_depends.repos` file.
- https://github.com/autowarefoundation/autoware.universe/blob/c9aa4093433f4e573c0898b393a5e98f883bdab7/.github/workflows/check-build-depends.yaml#L49C32-L49C51

### Expected behavior

clang-tidy-differential workflow should use the new repos files.

### Actual behavior

clang-tidy-differential workflow uses the old one.

### Steps to reproduce

Create a PR which has the workflow such as:
- https://github.com/autowarefoundation/autoware.universe/pull/10157

xmfcx, 2025-02-19:
- This was caused by https://github.com/autowarefoundation/autoware.universe/pull/10148

Will fix it as soon as possible.

xmfcx, 2025-02-19:
It is also confirmed in the reported PR too:
- https://github.com/autowarefoundation/autoware.universe/pull/10157

It is now resolved.

[image]

closed 2025-02-19 by xmfcx

mebasoglu, 2025-02-19:
Hello @xmfcx, thank you!