autowarefoundation/autoware_universe #10168: Ear clipping triangulation when loading .osm format map

seuganger, 2025-02-20:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

I have met a bug when loading lanelet2 format vector map when launching planning simulation. I made the vector map with bag2lanelet and Vector Map Builder Tool, and in the tool the vector map are what they expected to be. 

### Expected behavior

In the Vector Map Builder Tool, my vector map was like:

[image]
As it can be shown up in this tool, it should be used in simulation. 

### Actual behavior

When importing this map to planning simulation, it was like:

[image]
The white point in the middle of this picture is point cloud, and the green area is the lane area. It seem that a lanelet2 node was located far from other nodes. To be more specific, I will show pictures in different scales below. 

[image]

[image]

[image]

[image]

### Steps to reproduce

1. Use my lanelet2 and PCD files
https://entuedu-my.sharepoint.com/:f:/g/personal/n2402056d_e_ntu_edu_sg/EmqltdP8bZZHqo4XXkVgwOUBbUEuJiHzwt59VR5UaZrj5A?e=Qwj0EA
2. Download the files, rename them, put them in ~/autoware_map/rosbag-test (or your map path), change the launch file ~/autoware/src/launcher/autoware_launch/autoware_launch/launch/planning_simulator.launch.xml line 15,16 into map names
3. run the autoware planning simulation with
`ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/rosbag-test vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit`

### Versions

-OS: Ubuntu 22.04
-ROS: humble
-Autoware: universe (docker installation)
-Hardware: Nvidia 3090 GPU

### Possible causes

There is a warning: Could not find valid vertex for ear clipping triangulation. Triangulation result might be invalid. I suspect that a node was not initialized properly, which causes the ear clipping triangulation fail. It seems a node was put far away from its original place, which has been shown in the pictures above. I have checked all of my nodes in osm format lanelet map, none of them was far away from others such significantly to generate this problem in simulation. So the possible cause is, some node information was written in the code, when testing with sample maps, it did not arouse problems. 
Besides, I find out that the the distance between wrongly initialized node and point cloud is approximate the distance between sample map config and  my map config. This picture shows the distance between point cloud and wrong node:
[image]
The [lat, lon] in my config is: [latitude: 35.22312494103  longitude: 138.80245834626]
In sample config is: [latitude: 35.23808753540768  longitude: 139.9009591876285]
The distance between 2 map configs is about 60 kilometers, which echos the distance in the picture. 
So the conclusion is, the config file might be the cause of this problem, whose path is fixed in the code and the default is in sample-map-planning. In the next step, I will try to change the path of config file to find a solution. If it succeeded, I will report in the comments and give a suggestion on how to avoid this problem. However, this should still be reported as a bug because both the tutorial and the launch.xml file did not mention any information about the location of map config files.

YamatoAndo, 2025-02-25:
@seuganger Hi,
In `map_projector_info.yaml`, MGRS coordinate system `54SVE` is specified, but it seems that your map was not created using the MGRS coordinate system.
Please update `map_projector_info.yaml` as follows.

```
projector_type: Local
```

https://github.com/autowarefoundation/autoware.universe/tree/main/map/autoware_map_projection_loader#using-local-coordinate

seuganger, 2025-02-27:
Solved, thank you for your reply!

closed 2025-02-28 by YamatoAndo