autowarefoundation/autoware_universe #10176: Autoware_carla_interface does not work with CARLA

Sephiroth1118, 2025-02-20:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

I'm testing the co-simulation between autoware.universe and the CARLA simulator. I use the autoware_carla_interface. The launch file is `e2e_simulator.launch.xml`. Before this one, I ran the `planning_simulator.launch.xml` successfully.

This is the log file.

[launch.log](https://github.com/user-attachments/files/18882195/launch.log)
There are mainly three errors:

1. `[component_container_mt-74] [ERROR] [1739958327.899700638] [planning.scenario_planning.parking.costmap_generator]: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.`
2. `[service_log_checker_node-9] [ERROR] [1739958327.991541454] [system.service_log_checker]: /localization/initialize: status code 1 'The vehicle is not stopped.' (/localization/util/pose_initializer)`
3. `1740032228.9587953 [service_log_checker_node-9] [ERROR] [1740032228.957593316] [system.service_log_checker]: /localization/initialize: status code 4 'align server failed.' (/localization/util/pose_initializer)`

I searched the errors and found this issue: https://github.com/autowarefoundation/autoware.universe/issues/3208. I tried the methods mention in it but not work.

1. This the output of topic `/sensing/vehicle_velocity_converter/twist_with_covariance`

[image]
2. I then checked the `stop_check_enabled`, found it in the file `/home/qiujing/autoware/src/universe/autoware.universe/launch/tier4_simulator_launch/launch/simulator.launch.xml`, but it is already False.

[image]

### Expected behavior

I expected the "2D Pose Estimate" in Rviz will initialize the localization of the vehicle and then I can set the goal by "2D Pose Goal". 

### Actual behavior

Nothing happened. The localization of the vehicle can not be initialized.

### Steps to reproduce

1. Start the CARLA by `./CarlaUE4.sh`
2. Start autoware.universe by `ros2 launch autoware_launch e2e_simulator.launch.xml map_path:=$HOME/autoware_map/Town01 vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit simulator_type:=carla carla_map:=Town01`
3. Set the Pose of the vehicle.

### Versions

- OS: Ubuntu 22.04
- ROS2: Humble
- Autoware: autoware.universe from the `main` branch

### Possible causes

I think the vehicle is detected not stopped so that I can not set the Pose of the vehicle. But I do not know how to solve it. There are some similar problems from others, but not suitable for mine.

maxime-clem, 2025-02-20:
Do you have the same issue with AWSIM ? Your pointcloud container is crashing at launch but it is not clear why. How much video memory do you have available after running the Carla simulator ?

>
 ```
1740032149.5855720 [component_container_mt-5] [I] Loaded engine from /home/qiujing/autoware_data/lidar_centerpoint/pts_backbone_neck_head_centerpoint_tiny.engine
1740032149.6052928 [component_container_mt-5] [I] [TRT] [MemUsageChange] TensorRT-managed allocation in IExecutionContext creation: CPU +0, GPU +166, now: CPU 0, GPU 381 (MiB)
1740032149.6336644 [component_container_mt-5] *** Aborted at 1740032149 (unix time) try "date -d @1740032149" if you are using GNU date ***
```

Sephiroth1118, 2025-02-20:
Thank you so much for your reply!

1. Actually I haven't test it with AWSIM. I only test the CARLA.
2. I checked the video memory usage, it's 5148MiB /   6144MiB |, only 996 MiB is left. 

So this should be the core problem? My GPU is GTX 1660 SUPER, which is not a very good one today.

maxime-clem, 2025-02-20:
Yes 6GB is not enough to run both Autoware and Carla at the same time. Even with 8GB I sometimes have issues.
A possible workaround:
1. only run the `autoware_carla_interface` and the Carla simulator.
2. record a rosbag of the topics published by the interface.
3. close the Carla simulator, run Autoware.
4. replay the previously recorded bag (with the `--clock` argument).

This should allow you to at least test the perception and localization components.

Sephiroth1118, 2025-02-20:
Thank you so much! It seems like a feasible way.

By the way, may I ask whether there is anyway that do not use point cloud? I tried to delete the point cloud file when running `planning_simulator.launch.xml`. Although it will throw out the problem `1740039429.2630503 [component_container_mt-24] [pcl::PCDReader::read] Could not find file '/home/qiujing/autoware_map/sample-map-planning/pointcloud_map.pcd'.`, the function of autoware seems work well.

maxime-clem, 2025-02-20:
The pointcloud is needed for localization.

mcodinab, 2025-02-26:
Hello @Sephiroth1118, double check the network configuration.

Please, refer to [Localhost Settings](https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/) section.

It may help to link map with base_link. Let me know if it worked for you!

closed 2025-03-21 by idorobotics

maxime-clem, 2025-05-28:
This is your issue:
```
1748393177.4717185 [component_container_mt-5] [31m[ERROR 1748393177.470103784] [sensing.lidar.concatenate_data]: Cloud does not contain x, y, z fields. Cannot transform. (transformPointcloud() at /home/sora/Desktop/xiangl/autoware/src/universe/external/managed_transform_buffer/managed_transform_buffer/src/managed_transform_buffer.cpp:267)[0m
1748393177.4718804 [component_container_mt-5] [31m[ERROR 1748393177.470200537] [sensing.lidar.concatenate_data]: Concatenated cloud is an empty pointcloud. (publish_clouds() at /home/sora/Desktop/xiangl/autoware/src/universe/autoware_universe/sensing/autoware_pointcloud_preprocessor/src/concatenate_data/concatenate_and_time_sync_node.cpp:305)[0m
1748393177.4719362 [component_container_mt-5] [33m[WARN 1748393177.470286470] [sensing.lidar.concatenate_data]: transformed_raw_points[/sensing/lidar/right/pointcloud_before_sync] is nullptr, skipping pointcloud publish. (publish_clouds() at /home/sora/Desktop/xiangl/autoware/src/universe/autoware_universe/sensing/autoware_pointcloud_preprocessor/src/concatenate_data/concatenate_and_time_sync_node.cpp:348)[0m
1748393177.4719703 [component_container_mt-5] [33m[WARN 1748393177.470308480] [sensing.lidar.concatenate_data]: transformed_raw_points[/sensing/lidar/left/pointcloud_before_sync] is nullptr, skipping pointcloud publish. (publish_clouds() at /home/sora/Desktop/xiangl/autoware/src/universe/autoware_universe/sensing/autoware_pointcloud_preprocessor/src/concatenate_data/concatenate_and_time_sync_node.cpp:348)[0m
1748393177.4719949 [component_container_mt-5] [33m[WARN 1748393177.470703747] [perception.object_recognition.detection.centerpoint.lidar_centerpoint]: The compatible callback was called. This results in a performance loss. This behavior is probably not intended or a temporal measure (compatibleCallback() at /home/sora/Desktop/xiangl/autoware/src/universe/external/cuda_blackboard/src/cuda_blackboard_subscriber.cpp:121)[0m
1748393177.4720228 [component_container_mt-5] *** Aborted at 1748393177 (unix time) try "date -d @1748393177" if you are using GNU date ***
1748393177.4728823 [component_container_mt-5] [31m[ERROR 1748393177.472052129] [perception.obstacle_segmentation.crop_box_filter]: The pointcloud layout is not compatible with PointXYZIRCAEDT or PointXYZIRC. Aborting (faster_input_indices_callback() at /home/sora/Desktop/xiangl/autoware/src/universe/autoware_universe/sensing/autoware_pointcloud_preprocessor/src/filter.cpp:372)[0m
1748393177.4730022 [component_container_mt-5] [31m[ERROR 1748393177.472111013] [localization.util.crop_box_filter_measurement_range]: The pointcloud layout is not compatible with PointXYZIRCAEDT or PointXYZIRC. Aborting (faster_input_indices_callback() at /home/sora/Desktop/xiangl/autoware/src/universe/autoware_universe/sensing/autoware_pointcloud_preprocessor/src/filter.cpp:372)[0m
1748393177.4762323 [component_container_mt-5] PC: @                0x0 (unknown)
1748393177.5114636 [component_container_mt-5] *** SIGFPE (@0x7df9bf724ec7) received by PID 3894864 (TID 0x7dfa677fe640) from PID 18446744072626523847; stack trace: ***
1748393177.5128193 [component_container_mt-5] [0m[INFO 1748393177.512386867] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<autoware::compare_map_segmentation::CompareElevationMapFilterComponent> (create_component_factory() at ./src/component_manager.cpp:131)[0m
1748393177.5135052 [component_container_mt-5] [0m[INFO 1748393177.512454054] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<autoware::compare_map_segmentation::DistanceBasedCompareMapFilterComponent> (create_component_factory() at ./src/component_manager.cpp:131)[0m
1748393177.5140676 [component_container_mt-5] [0m[INFO 1748393177.512481887] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<autoware::compare_map_segmentation::VoxelBasedApproximateCompareMapFilterComponent> (create_component_factory() at ./src/component_manager.cpp:131)[0m
1748393177.5152915 [component_container_mt-5] [0m[INFO 1748393177.512493313] [pointcloud_container]: Found class: rclcpp_components::NodeFactoryTemplate<autoware::compare_map_segmentation::VoxelBasedCompareMapFilterComponent> (create_component_factory() at ./src/component_manager.cpp:131)[0m
1748393177.5154667 [component_container_mt-5] [0m[INFO 1748393177.512502508] [pointcloud_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<autoware::compare_map_segmentation::VoxelBasedCompareMapFilterComponent> (create_component_factory() at ./src/component_manager.cpp:133)[0m
1748393177.5176656 [component_container_mt-5]     @     0x7dfa743ab4d6 google::(anonymous namespace)::FailureSignalHandler()
1748393177.5243979 [component_container_mt-5]     @     0x7dfa76842520 (unknown)
1748393177.5270441 [component_container_mt-5]     @     0x7df9bf724ec7 CudaPointCloud2::fromROSMsgAsync()
1748393177.5292535 [component_container_mt-5]     @     0x7df9bf70cd14 autoware::occupancy_grid_map::PointcloudBasedOccupancyGridMapNode::rawPointcloudCallback()
1748393177.5305665 [component_container_mt-5]     @     0x7df9bf73cff8 rclcpp::experimental::SubscriptionIntraProcess<>::execute_impl<>()
1748393177.5342457 [component_container_mt-5]     @     0x7dfa76fcb19f rclcpp::Executor::execute_any_executable()
1748393177.5370789 [component_container_mt-5]     @     0x7dfa76fd249a rclcpp::executors::MultiThreadedExecutor::run()
[... cut ...]
1748393177.8024065 [ERROR] [component_container_mt-5]: process has died [pid 3894864, exit code -8, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=pointcloud_container -r __ns:=/ -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].
```
I am not sure what could be the problem.
It may be worth opening a new issue.