autowarefoundation/autoware_universe #10230: False positive of emergency stop while going downhill.

Kim-mins, 2025-03-05:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

Hi team,

I'm currently running Autoware in Carla, and I found a situation that Autoware makes false positive of emergency stop while going downhill.

Here's a video of the situation: [[rviz](https://salbox.synology.me/drive/d/s/12PTeDUVaHIRbiyk8TdgbqeVlJrAqgv8/gsmy-BjdMAUfQqNe9Zbcg9Ue4kWNaA0s-WL2gXEHBGQw)]

At 0:20 of the video, there's an emergency stop once when started to drive down, and it makes the vehicle stop, and the vehicle does not drive at all.

[image]

- After multiple tests, I found that this issue occurs **only when the ego vehicle is initialized near a downhill**. Simply driving downhill does not trigger this issue.
- I also tried the solution of the similar issue [here](https://github.com/autowarefoundation/autoware.universe/issues/9672), but it did not work.
- For the stuck after an emergency stop, I guess [this issue](https://github.com/autowarefoundation/autoware.universe/issues/9666) is the cause.
- The shape of drivable area seems weird, but I checked that the shape of the lane in my current map is correct. [[map file](https://drive.google.com/drive/folders/1rt1pKEfY-hqC2pMrHfajg1EVfjlIcAX5)]

### Expected behavior

I hope the ego vehicle to drive downhill successfully.

### Actual behavior

But the ego vehicle stuck in the middle of a downhill.

### Steps to reproduce

Please replay the [ros2bag file](https://salbox.synology.me/drive/d/s/12PTh4gB8NHzf8iKZ8BKGZsunNWqlJZL/z8EOsdlKnFm-OAaLWxyyA12MIY3Fi-qy-zrTg0O7EGQw).

### Versions

- OS: Ubuntu 22.04 (Docker), Ubuntu 20.04 (host)
- Autoware: [0.39.1](https://github.com/autowarefoundation/autoware/tree/0.39.1)
- Carla: 0.9.15
- autoware_carla_interface: [link](https://github.com/autowarefoundation/autoware.universe/tree/0.39.0/simulator/autoware_carla_interface)

Kim-mins, 2025-03-06:
Thank you for the response and the suggestion for debugging @takayuki5168!!

I checked the log from `diagnostic_graph_utils`, and I observed that some warnings are there.

<details>
<summary>Here's a snapshot of the result</summary>

```
| No | Level | Path                                                          | Type              | Children                     |
|----|-------|---------------------------------------------------------------|-------------------|-------------------------------||----|-------|----------------------------------|------|----------||----|-------|---------------------------------------------------------------|-------------------|------------------------------|
|  1 | OK    | /autoware/modes/stop                                          | ok                |                              |
| 14 | WARN  | /autoware/modes/autonomous                                    | and               | 5 4 2 20 6 24 23             |
| 15 | OK    | /autoware/modes/local                                         | and               | 24 23 17                     |
| 16 | OK    | /autoware/modes/remote                                        | and               | 24 23 25                     |
| 17 | OK    | /autoware/modes/emergency_stop                                | and               | 24 23                        |
| 18 | WARN  | /autoware/modes/comfortable_stop                              | and               | 5 4 2 20 6 24 23             |
| 19 | WARN  | /autoware/modes/pull_over                                     | and               | 5 4 2 20 6 24 23             |
| 13 | OK    | /autoware/debug/tools                                         | and               | 55                           |
|  5 | OK    | /autoware/map                                                 | and               | 27 22                        |
|  4 | WARN  | /autoware/localization                                        | short-circuit-and | 23 3                         |
|  3 | WARN  |                                                               | and               | 24 25 21 28                  |
|  2 | OK    | /autoware/planning                                            | short-circuit-and | 29 1                         |
|  1 | OK    |                                                               | and               | 30 31 7                      |
|  7 | OK    | /autoware/planning/trajectory_validation                      | and               | 32 33 34 35 7 37 38 39 40 41 |
| 20 | OK    | /autoware/perception                                          | and               | 42 43                        |
|  6 | OK    | /autoware/control                                             | and               | 44 45 46 45 48 49 50         |
| 17 | OK    | /autoware/control/local                                       | and               | 51 52                        |
| 25 | OK    | /autoware/control/remote                                      | and               | 51 52                        |
| 24 | OK    | /autoware/vehicle                                             | and               | 53 54                        |
| 23 | OK    | /autoware/system                                              | and               | 56 57                        |
| 27 | OK    | /autoware/map/topic_rate_check/vector_map                     | diag              |                              |
| 22 | OK    | /autoware/map/topic_rate_check/pointcloud_map                 | diag              |                              |
| 23 | OK    | /autoware/localization/state                                  | diag              |                              |
| 24 | OK    | /autoware/localization/topic_rate_check/transform             | diag              |                              |
| 25 | OK    | /autoware/localization/topic_rate_check/pose_twist_fusion     | diag              |                              |
| 22 | OK    | /autoware/localization/scan_matching_status                   | diag              |                              |
| 21 | OK    | /autoware/localization/accuracy                               | diag              |                              |
| 28 | WARN  | /autoware/localization/sensor_fusion_status                   | diag              |                              |
| 29 | OK    | /autoware/planning/routing/state                              | diag              |                              |
| 30 | OK    | /autoware/planning/topic_rate_check/route                     | diag              |                              |
| 31 | OK    | /autoware/planning/topic_rate_check/trajectory                | diag              |                              |
| 32 | OK    | /autoware/planning/trajectory_validation/finite               | diag              |                              |
| 33 | OK    | /autoware/planning/trajectory_validation/interval             | diag              |                              |
| 34 | OK    | /autoware/planning/trajectory_validation/curvature            | diag              |                              |
| 35 | OK    | /autoware/planning/trajectory_validation/angle                | diag              |                              |
|  7 | OK    | /autoware/planning/trajectory_validation/lateral_acceleration | diag              |                              |
[... cut ...]
| 55 | OK    | /autoware/system/service_log_checker                          | diag              |                              |
| 56 | OK    | /autoware/system/topic_rate_check/emergency_control_command   | diag              |                              |
| 57 | OK    | /autoware/system/emergency_stop_operation                     | diag              |                              |
```

</details>

It seems that `/autoware/localization/sensor_fusion_status` has an issue, but even after removing that diagnostic, the vehicle still cannot drive at all.. Could you please share your idea?

Thank you!

takayuki5168, 2025-03-06:
@Kim-mins 
If the MRM still happened after you removed the diagnostic, what was the result of the `diagnostic_graph_utils`'s log?