autowarefoundation/autoware_universe #10275: How to run the Yabloc localization

GPrathap, 2025-03-14:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

### Contents

I want to run the Yabloc localization 

Instructions provided here: https://autowarefoundation.github.io/autoware.universe/main/localization/yabloc/  seems to be wrong, 

I've found another bag file: nishishinjuku_autoware_map

### Expected behavior

But I can't see the vehicle moving. :) Could someone please tell me what steps I need to take to get Yabloc working?

I built the autoware docker image based on `autoware-nightly.repos`

### Actual behavior

Could not produce result as https://autowarefoundation.github.io/autoware.universe/main/localization/yabloc/

### Steps to reproduce

`ros2 launch autoware_launch logging_simulator.launch.xml map_path:=$AUTOWARE_HOME/autoware_map/nishishinjuku_autoware_map vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit pose_source:=yabloc`

Then I ran this bag file:

`ros2 bag play /home/autoware/data/autoware_map/sample_bag/yabloc/input_bag`

### Versions

Ubuntu 22.04 + ROS2 humble

YamatoAndo, 2025-03-17:
same issue https://github.com/orgs/autowarefoundation/discussions/5883

YamatoAndo, 2025-03-17:
@KYabuuchi Could you answer this issue?

KYabuuchi, 2025-03-17:
@GPrathap  Hi. Which ROSBAG are you using?  

There are several public ROSBAG available for running YabLoc:

1. [ROSBAG from Arbiter README](https://drive.google.com/file/d/1ZNlkyCtwe04iKFREdeZ5xuMU_jWpwM3W/view?usp=sharing)  
2. [ROSBAG from Driving Log Replayer](https://docs.google.com/uc?export=download&id=17ppdMKi4IC8J_2-_9nyYv-LAfW0M1re5)  

Please note that some public ROSBAGs are outdated due to updates in Autoware msg types.  
For example,    [Obsolete ROSBAG (linked from the yabloc introducing PR)](https://drive.google.com/file/d/1zKTGRH4lD-wptpOdNCgpiPfGRDP0XrUm/view?usp=sharing)  

Since the maps are compatible, you can use any of nishishinjuku map.  

---
Also, please verify that the **ROSBAG contains the `/clock` message**.  
If it does not, make sure to play the ROSBAG with the `--clock` option:  

```sh
rosbag play your_rosbag.bag --clock
```
Some public ROSBAGs include `/clock` , while others do not.

shpark98, 2025-03-18:
@KYabuuchi  

I'm trying to run YabLoc in a real-world environment.

Does the PCD map need to include GPS information?

Since I'm planning to proceed without LiDAR, I'm curious about how YabLoc determines or matches the initial position.

GPrathap, 2025-03-18:
Hi @KYabuuchi , 

Finally, these two commands work:

ros2 launch autoware_launch logging_simulator.launch.xml map_path:=$AUTOWARE_HOME/autoware_map/nishishinjuku_autoware_map   vehicle_model:=sample_vehicle   sensor_model:=awsim_sensor_kit   control:=false   planning:=false   pose_source:=yabloc

ros2 bag play ~/autoware_map/sample_bag/yabloc/input_bag --clock 

Maybe update https://autowarefoundation.github.io/autoware.universe/main/localization/yabloc/ so that someone can run it without any problem , thank youuu

KYabuuchi, 2025-03-18:
@shpark98  YabLoc determines its initial position using **GNSS** and estimates orientation by matching semantic segmentation results with a vector map.  
Without GNSS, the initial position cannot be automatically determined.  

In environments where GNSS is unavailable, such as indoors, global coordinate information in the PCD is not necessary.
In such cases, GNSS-based position correction and automatic initial position estimation is not available.

If you plan to use it outdoors, I recommend creating the PCD in a global coordinate system.
For example, the sample data I am proposing in this thread uses global coordinates (MGRS).

shpark98, 2025-03-18:
Thanks for your reply @KYabuuchi 

I understand that using GNSS as the initial position means comparing the current GNSS location with something. 
Is my understanding correct?

If so, does the GNSS information exist within the point cloud itself? 
Or is it embedded in the vector map created by the vector map builder?

KYabuuchi, 2025-03-18:
@shpark98 GNSS positions are not compared to anything. Autoware projects the latitude and longitude output by GNSS into local coordinates using the specified projection type.

Autoware supports several projection methods for converting latitude/longitude to local coordinates, and [this page](https://github.com/autowarefoundation/autoware.universe/tree/main/map/autoware_map_projection_loader#map-projector-info-file-specification) provides detailed information.
For example, if using MGRS, latitude = 36° and longitude = -115° would be projected to `11SPV 67305, 97905`.

If you use the MGRS projection type, you need to create the PCD's x, y coordinates in the MGRS XXXXX, YYYYY coordinate system.

[image]

shpark98, 2025-03-18:
@KYabuuchi I'm glad given image helped. Your answers have been extremely helpful!

So, if you transform the origin of the PCD map based on the MGRS information, the GNSS data will be aligned with the PCD map, allowing you to use GNSS within Autoware.

This is just a personal curiosity, but do you know the approximate error if I only connect GNSS to Autoware without using Yabloc?
And if the vector map only contains roads without additional information like traffic lights, can Yabloc still function properly?

KYabuuchi, 2025-03-18:
@shpark98 
> t do you know the approximate error if I only connect GNSS to Autoware without using Yabloc?
I'm not sure.

>  if the vector map only contains roads without additional information like traffic lights, can Yabloc still function properly?
Since YabLoc relies on road markings, information other than the road surface is unnecessary.
YabLoc functions even if the map does not include traffic signals.

GPrathap, 2025-03-19:
Hi @KYabuuchi , 

I have a question regarding  how to build a cost map, 

1. It uses line detection on the image and gets the point cloud whose unit is pixles
 
https://github.com/autowarefoundation/autoware_universe/blob/main/localization/yabloc/yabloc_particle_filter/src/camera_corrector/camera_particle_corrector_core.cpp#L60

2. Lanelet extraction of lane marking are in the actual coordinate system, 

https://github.com/autowarefoundation/autoware_universe/blob/main/localization/yabloc/yabloc_particle_filter/src/camera_corrector/camera_particle_corrector_core.cpp#L62

How does this HierarchicalCostMap and direct_cost_map work?

YamatoAndo, 2025-05-19:
@KYabuuchi Could you answer the question?

KYabuuchi, 2025-05-19:
@GPrathap  
> It uses line detection on the image and gets the point cloud whose unit is pixles
main/localization/yabloc/yabloc_particle_filter/src/camera_corrector/camera_particle_corrector_core.cpp#L60

The line_segments_cloud is a set of line segments projected onto the ground in real-world coordinates. The unit is not pixels.
The line segments extracted from the image are projected onto the ground within the segment_filter.

[image]

> Can I know what kind of modification I have to do HierarchicalCostMap?

I do not recommend editing HierarchicalCostMap because the module is complex.
It’s better to simulate its input, `ll2_road_marking` which is a set of line segments.
I think the easiest approach is to implement a node that generates it from a point cloud, following the logic of ll2_decomposer.

[image]

KYabuuchi, 2025-05-19:
> How to run the Yabloc localization #10275

Also, since the original issue mentioned in the title has already been resolved, I will close this issue.

If you have any problems regarding customization, please feel free to open a new issue.

closed 2025-05-19 by KYabuuchi

GPrathap, 2025-05-19:
@KYabuuchi  thank youu