autowarefoundation/autoware_universe #10285: Drivable area limitation -before red traffic light- fails when driving in narrow lane

MertClk, 2025-03-17:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

The avoidance module considers traffic light information to determine the appropriate behavior for the ego vehicle. If the ego has not yet initiated an avoidance maneuver, the module restricts the maximum shift length and ensures that the vehicle remains **only** within the current lane during a red traffic signal. This prevents the ego from obstructing other vehicles, even if it begins an avoidance maneuver but subsequently stops at a red signal.

However, when the ego vehicle is driving in a narrow lane, and its footprint exceeds the boundaries of the concatenated lanelets (previous, current, and next), the avoidance module **fails** to enforce drivable area restrictions. This results in unintended behavior.

A video demonstrating the issue is provided below.

[Demonstration](https://youtu.be/VcYZDIpdtjc)

### Expected behavior

- The ego behaves as documented in the [readme](https://github.com/autowarefoundation/autoware.universe/tree/main/planning/behavior_path_planner/autoware_behavior_path_static_obstacle_avoidance_module#planning-at-red-traffic-light) 

### Actual behavior

- The ego vehicle starts avoidance maneuver and blocks the adjacent lane 

### Steps to reproduce

This issue can be reproduced with the provided scenario file and Autoware sample map which can be download from [here](https://drive.google.com/file/d/1499_nsbUbIeturZaDj7jhUownh5fvXHd/view).

### Possible causes

[This](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/autoware_behavior_path_static_obstacle_avoidance_module/src/scene.cpp#L221) value returns false, since ego exceeds the the boundary while turning.

MertClk, 2025-03-17:
@satoshi-ota @yhisaki This issue impacts the avoidance module's behavior in narrow lanes. When you have a moment, could you kindly share your thoughts on this?

yhisaki, 2025-07-10:
solved in this PR
- https://github.com/autowarefoundation/autoware_universe/pull/10948
Thank you.

closed 2025-07-10 by yhisaki