autowarefoundation/autoware_universe #10287: Static Obstacle Avoidance fails to avoid parked vehicle after limited avoid_margin

MertClk, 2025-03-17:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

Objects without an avoid_margin are not considered is_avoidable. As a result, the static_obstacle_avoidance module transitions to the onExit process, ultimately causing the ego vehicle to move and miss the deadline."

[Video](https://youtu.be/MhiZ8nllamY)

### Expected behavior

- Ego waits until it can avoid the parked vehicle.
- Starts avoidance when the space is empty.

### Actual behavior

- Ego drives toward the parked vehicle and overran the deadline.

### Steps to reproduce

A scenario file has been generated for this issue. Please use following map and scenario file.

Map: [Drive](https://drive.google.com/file/d/1IniR5QFkE8Naum_i5Mn6MlXPGfsAnLRO/view?usp=sharing)
Scenario File: 

### Possible causes

In shif_line_generator, when the drivable area is limited temporarily, the target object is not kept as avoidable.
[shift_line_generator.cpp@274](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/autoware_behavior_path_static_obstacle_avoidance_module/src/shift_line_generator.cpp#L274)

```    if (!o.avoid_margin.has_value()) { 
      if (
        data.red_signal_lane.has_value() &&
        data.red_signal_lane.value().id() == o.overhang_lanelet.id()) {
        o.info = ObjectInfo::LIMIT_DRIVABLE_SPACE_TEMPORARY;
      } else {
        o.info = ObjectInfo::INSUFFICIENT_DRIVABLE_SPACE;
      }
      if (o.avoid_required && is_forward_object(o) && is_on_path(o)) {
        break;
      } else {
        unavoidable_objects.push_back(o);
        continue;
      }
    }
 ```
 For this reason, the object becomes _isAbsolutelyNotAvoidable_  and this will cause static_obstacle_avoidance module to move to **onExit** phase.
[scene.cpp@564](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/autoware_behavior_path_static_obstacle_avoidance_module/src/scene.cpp#L564)

As a solution, the object should become "is_avoidable" before following [break](https://github.com/autowarefoundation/autoware.universe/blob/main/planning/behavior_path_planner/autoware_behavior_path_static_obstacle_avoidance_module/src/shift_line_generator.cpp#L275) statement.

[Solution](https://youtu.be/EzveylMVKZo)

MertClk, 2025-03-17:
@satoshi-ota @yhisaki  This issue impacts the avoidance module's behavior in narrow lanes. When you have a moment, could you kindly share your thoughts on this?

satoshi-ota, 2025-04-02:
@MertClk I met similar issue and found a bug of avoidance module.
I've fixed it in https://github.com/autowarefoundation/autoware_universe/pull/10242 and I guess this PR effective for your issue as well. Could you try to apply above PR on your branch?

MertClk, 2025-04-02:
@satoshi-ota 
Thank you so much for your follow up on this issue. I tried this scenario with the latest Autoware and I confirm that I'm using [#10242](https://github.com/autowarefoundation/autoware_universe/pull/10242). Here are my findings,

1. The default `stop_buffer` value is set to 1 m. With this value, the scenario fails and ego can not keep stopping.

https://github.com/user-attachments/assets/2eb48784-14a6-4f9d-ad44-c7691f86bf81

2. When `stop_buffer` set to 2 m, it is possible to pass the scenario. However, ego still can not keep stopping. 

https://github.com/user-attachments/assets/4d7853ee-261f-40d8-bc2a-0e6eb36c3f87

3. If we modify the scenario so that the adjacent lane is busy a bit more after the traffic light turns green, the scenario fails again. 

https://github.com/user-attachments/assets/98e4beff-963a-4be0-a433-f264a7765971

[New scenario file](https://drive.google.com/file/d/1L2eMBkVwT5m1YglyqhInsEGhK92l99oc/view?usp=sharing)

satoshi-ota, 2025-07-09:
Sorry for the late reply. 

> Is left-lane avoidance from the right lane currently supported in the static obstacle avoidance module?

Left-lane avoidance from the right lane is supported.

However, please note that the avoidance module does not attempt to avoid every stopped object. It is designed to avoid objects that are not expected to move again in the short term.

For example, parked vehicles are assumed to remain stationary for a while, so the module determines that they should be avoided.
On the other hand, for vehicles stopped at a red traffic light, the ego vehicle can simply continue following them once the light turns green, so avoidance is not strictly necessary. As a result, the current implementation does not perform avoidance in such cases.