autowarefoundation/autoware_universe #10317: Centerpoint cannot detect any objects for livox mid 360

leoxiang66, 2025-03-21:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

Hi guys, I tried both `centerpoint` and `centerpoint_tiny` models but both cannot detect any objects.

I see on [this issue](https://github.com/autowarefoundation/autoware_universe/issues/5362) that the training data matters (e.g. different beams or scan mode).

I would ask to see if anyone has relevant experience on centerpoint for mid 360?

Many thanks!

### Expected behavior

detecting objects

### Actual behavior

detect nothing

### Steps to reproduce

normal steps

### Versions

ubuntu2204
ros2 humble

knzo25, 2025-03-24:
@leoxiang66 
If centerpoint can not detect "any" objects it is probably not due to the lidar itself but rather data format, environment it was tested and so on. If you can provide a rosbag, we can check it later

leoxiang66, 2025-03-24:
@knzo25 
Hi Knzo25, thanks for reply!

You can find the rosbag here: https://drive.google.com/file/d/1y3UIEVVIhrxvtQM4CJ7HMVeFMlaBl3od/view?usp=sharing

Btw, I have also tested inferencing on the original training dataset (nuscenes), and centerpoint can detect objects. So I doubt it may be the different distribution of these two datasets.

May you have a test on this rosbag too? I would appreciate it!