autowarefoundation/autoware_universe #10366: Failed to give three or more times "2D Goal Pose" in simulation

Leslie-Jane, 2025-03-28:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

It was collapsed when I clicked and published two or three times "2D Goal Pose" after setting a "2D Pose Estimate"。Is it caused by planning ? 

[image]

### Expected behavior

Please tell me why and how to solve this issue.

### Actual behavior

I cann't do anything after publish two or three times "2D Goal Pose" unless the program is closed and relaunched.

### Steps to reproduce

1. Execute program “ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/fulongma vehicle_model:=private_hw_vehicle sensor_model:=private_hw_sensor_kit”
2. Click "2D Pose Estimate" and click "2D Goal Pose" for one time
3. Change "2D Goal Pose" position for more times

### Versions

ROS2 humble

maxime-clem, 2025-03-28:
Can you share the `launch.log` file (found in `HOME/.ros/log/{DATE-TIME-...}/`) when the issue occur ?
It sounds like a known issue where some node crashes when sending a new goal and we can make sure by looking at the logs.

Leslie-Jane, 2025-03-28:
> Can you share the `launch.log` file (found in `HOME/.ros/log/{DATE-TIME-...}/`) when the issue occur ? It sounds like a known issue where some node crashes when sending a new goal and we can make sure by looking at the logs.

[image]
Thanks sir,You replied too promptly！I found error in the terminal above

Leslie-Jane, 2025-03-28:
> Can you share the `launch.log` file (found in `HOME/.ros/log/{DATE-TIME-...}/`) when the issue occur ? It sounds like a known issue where some node crashes when sending a new goal and we can make sure by looking at the logs.

Sure

Leslie-Jane, 2025-03-28:
> Can you share the `launch.log` file (found in `HOME/.ros/log/{DATE-TIME-...}/`) when the issue occur ? It sounds like a known issue where some node crashes when sending a new goal and we can make sure by looking at the logs.

Sure

[launch.log](https://github.com/user-attachments/files/19497043/launch.log)

maxime-clem, 2025-03-28:
Thank you. This looks like https://github.com/autowarefoundation/autoware/issues/4138
You can see a possible solution here (build `rclcpp` locally to include some needed bug fixes): https://github.com/autowarefoundation/autoware/issues/4138#issuecomment-2264853842

Leslie-Jane, 2025-03-28:
> Thank you. This looks like [autowarefoundation/autoware#4138](https://github.com/autowarefoundation/autoware/issues/4138) You can see a possible solution here (build `rclcpp` locally to include some needed bug fixes): [autowarefoundation/autoware#4138 (comment)](https://github.com/autowarefoundation/autoware/issues/4138#issuecomment-2264853842)

Thanks sir

closed 2025-07-07 by maxime-clem