autowarefoundation/autoware_universe #10410: CMake Error at `/opt/ros/humble/share/tinyxml2_vendor/cmake/tinyxml2_vendor-extras.cmake`

youtalk, 2025-04-08:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

By removing the dependency from `autoware_lanelet2_extension` to `autoware_utils`, an error has started to occur in various packages in `autoware_universe` indicating that `tinyxml2` cannot be found. 

https://github.com/autowarefoundation/autoware_lanelet2_extension/pull/47

This issue likely existed before but went unnoticed because it was being indirectly resolved through `autoware_utils`, which apparently handled the `tinyxml2` dependency implicitly.

### Expected behavior

No `tinyxml2` error occurs.

### Actual behavior

```
#109 18.23 CMake Error at /opt/ros/humble/share/tinyxml2_vendor/cmake/tinyxml2_vendor-extras.cmake:23 (message):
#109 18.23   Unable to extract the library file path from
#109 18.23 Call Stack (most recent call first):
#109 18.23   /opt/ros/humble/share/tinyxml2_vendor/cmake/tinyxml2_vendorConfig.cmake:41 (include)
#109 18.23   /opt/ros/humble/share/pluginlib/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
#109 18.23   /opt/ros/humble/share/pluginlib/cmake/pluginlibConfig.cmake:41 (include)
#109 18.23   /opt/ros/humble/share/ament_cmake_auto/cmake/ament_auto_find_build_dependencies.cmake:67 (find_package)
#109 18.23   /opt/autoware/share/autoware_cmake/cmake/autoware_package.cmake:49 (ament_auto_find_build_dependencies)
#109 18.23   CMakeLists.txt:5 (autoware_package)
``

### Steps to reproduce

Build `autoware_universe` using `autoware_lanelet2_extension` version 0.7.0.
https://github.com/autowarefoundation/autoware_lanelet2_extension/releases/tag/0.7.0

youtalk, 2025-04-08:
We were discussing a workaround in https://github.com/autowarefoundation/autoware_cmake/pull/24. A fundamental solution might require modifications to either `tinyxml2_vendor` or `ament_cmake_auto`.

mitsudome-r, 2025-04-08:
@esteve Will check if there are better solution to the issue.

esteve, 2025-11-25:
https://github.com/autowarefoundation/autoware_cmake/pull/30 and https://github.com/tier4/scenario_simulator_v2/pull/1604 can now be merged and this issue can be closed afterwards