autowarefoundation/autoware_universe #10427: Ego frame ID is hard-coeded in "autoware_multi_object_tracker"

urasakikeisuke, 2025-04-09:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

We are trying to use autoware_multi_object_tracker in a system that does not have "base_link" as an ego frame. However, as shown below, since the "base_link" is hard-coded as an ego frame, nothing is output from the tracker because it is always rejected in the nearest point calculation part.

https://github.com/autowarefoundation/autoware_universe/blob/a20e0e6ec40577b5f66d7a994ff60a1d68332a26/perception/autoware_multi_object_tracker/include/autoware/multi_object_tracker/odometry.hpp#L54

https://github.com/autowarefoundation/autoware_universe/blob/a20e0e6ec40577b5f66d7a994ff60a1d68332a26/perception/autoware_multi_object_tracker/src/processor/input_manager.cpp#L98-L102

I think it should be possible to change the ego frame from the configuration file.

### Expected behavior

The results are output even if the ego frame is not “base_link”.

### Actual behavior

Nothing is output if the ego frame is not “base_link”.

### Steps to reproduce

Run autoware_multi_object_tracker on data with non “base_link” ego frames in the default configuration.

### Versions

v0.43.0

amadeuszsz, 2025-04-10:
Hi @urasakikeisuke 

Thank you for reporting this issue. I addressed your idea in https://github.com/autowarefoundation/autoware_universe/pull/10428 and https://github.com/autowarefoundation/autoware_launch/pull/1397.
Let's check if we can merge it.

urasakikeisuke, 2025-04-10:
Hi @amadeuszsz 

Thank you for the quick response, I haven't tested autoware_launch as I don't use it, but I had no problem with the node. Thank you very much.

closed 2025-04-10 by amadeuszsz