autowarefoundation/autoware_universe #10470: Replace autoware_utils in simulator component

isamu-takagi, 2025-04-16:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

Replace `autoware_utils` with the split `autoware_utils_*` package.

### Purpose

Remove unnecessary dependencies.

### Possible approaches

Refer to https://github.com/autowarefoundation/autoware_utils/issues/34 for the replacement package.

### Definition of done

- [ ] autoware_universe/simulator/autoware_dummy_perception_publisher
- [ ] autoware_universe/simulator/autoware_simple_planning_simulator
- [ ] autoware_universe/simulator/autoware_fault_injection

kazu-321, 2025-04-21:
I will do `autoware_universe/simulator/autoware_dummy_perception_publisher`

kazu-321, 2025-04-21:
PR(`autoware_universe/simulator/autoware_dummy_perception_publisher`): https://github.com/autowarefoundation/autoware_universe/pull/10519

I will do `autoware_universe/simulator/autoware_simple_planning_simulator` next.

kazu-321, 2025-04-21:
PR(`autoware_universe/simulator/autoware_simple_planning_simulator`): https://github.com/autowarefoundation/autoware_universe/pull/10520
I will do `
autoware_universe/simulator/autoware_fault_injection` next.

kazu-321, 2025-04-21:
PR(`autoware_universe/simulator/autoware_fault_injection`): https://github.com/autowarefoundation/autoware_universe/pull/10521

closed 2025-10-16 by YamatoAndo