autowarefoundation/autoware_universe #10481: LiDAR semantic segmentation for Autoware

amadeuszsz, 2025-04-17:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

Introduce a new module for LiDAR semantic segmentation to the perception component. This module will classify 3D LiDAR point clouds into semantic categories such as road, sidewalk, vehicle, pedestrian, vegetation, etc., using a machine learning model. The module should process raw point cloud data and produce per-point semantic labels.

### Purpose

* Enhance perception capability with richer understanding of the scene.
* Improve robustness in complex environments where ground segmentation alone is insufficient.

### Possible approaches

* Integrate SOTA algorithms into Autoware.
  * [FRNet](https://arxiv.org/abs/2312.04484)
  * [Point Transformer V3](https://arxiv.org/abs/2312.10035)

### Definition of done

- [ ] Selected and integrated a suitable semantic segmentation ML model into [AWML](https://github.com/tier4/AWML?tab=readme-ov-file#3d-segmentation) with one of available OSS dataset for 3D semantic segmentation.
  - [ ] FRNet 
  - [ ] Point Transformer V3
- [ ] Implemented a Python inference PoC of processing pipeline to take raw point cloud input and output semantic labels.
  - [ ] FRNet 
  - [ ] Point Transformer V3
- [ ] Integrated module as a ROS C++ node with proper message interfaces and topics.
  - [ ] FRNet
  - [ ] Point Transformer V3
- [ ] Retrained model with internal dataset.
  - [ ] FRNet
  - [ ] Point Transformer V3

amadeuszsz, 2025-04-18:
PR for FRNet in https://github.com/autowarefoundation/autoware_universe/pull/10503.

knzo25, 2025-05-12:
universe's PR for PTv3 in https://github.com/autowarefoundation/autoware_universe/pull/10600

In addition, the following tensorrt plugins are needed:
 - https://github.com/autowarefoundation/autoware_universe/pull/10601
 - https://github.com/autowarefoundation/autoware_universe/pull/10602
 - https://github.com/autowarefoundation/autoware_universe/pull/10603
 - https://github.com/autowarefoundation/autoware_universe/pull/10604

knzo25, 2025-10-04:
@amadeuszsz 
All PRs are merged. We good to close this?