autowarefoundation/autoware_universe #10596: lane change module gives inconsistent result in specific situation

Joonki0111, 2025-05-08:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

When there is a goal in front of a obstacle and lane change is also needed to be happened, the lane change module gives a inconsistent result as shown like the footage below.

https://github.com/user-attachments/assets/66fe2d42-828f-4981-bf5d-caa1a5ad1422

It sometimes successfully reaches to the goal but sometimes get's stuck behind the obstacle.

https://github.com/user-attachments/assets/c6383678-4096-49e1-8004-8f75a2c7fa2f

I have changed some parameters like `prepare_duration`, `lateral_acceleration.max_values: [1.0, 1.0, 1.0]`.
Is this a known issue? or is it because I'm using a older version of Autoware?

### Expected behavior

Reach to the goal by lane change

### Actual behavior

Randomly behaves which might lead to failure.

### Steps to reproduce

simple planning simulator

### Versions

Ubuntu 22.04
Humble
release/v1.0

zulfaqar-azmi-t4, 2025-05-09:
Thank you so much for the report.
Case 1 is not so clear to me. Do you mean ego couldn't lane change properly?

In case 2. The target object is too close to the goal. In lane change, we compute the minimum lane change distance, and in your case, the default parameter is not suitable. You can try increasing lateral jerk to -1.0 m/s3, and minimum lane change velocity to 1.5m/s.

Joonki0111, 2025-05-09:
@zulfaqar-azmi-t4 
Case 1 & 2 has the same issue, which is giving two different paths rapidly. And this is what I described as 'inconsistent result'.
This result leads to random arrival. It could arrive as normal(not shown at the video), or it could just get stuck behind obstacle.

Thanks for your reply :)

zulfaqar-azmi-t4, 2025-06-18:
@Joonki0111 
Most probably this happens in old version.
With the latest version, the result shouldn't flicker that much and it has a more stable result.

Also I noticed that the stop line at the end of the trajectory seems to be short, did you happened to modify minimum lane change velocity?

Joonki0111, 2025-07-17:
@zulfaqar-azmi-t4 Sorry for the late reply, yes I have made some changes in lane change parameters. Currently, this is not a issue anymore because we made it through with this situation. I also tried the latest version of Autoware and seems good enough. Thanks :)

maxime-clem, 2025-07-23:
From the last reply it seems the issue is solved. If not please reopen the issue.