autowarefoundation/autoware_universe #10610: CUDA related libraries are not being compiled for target architectures

knzo25, 2025-05-12:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

While developing the ptv3 node I realized that cuda code does is not being compiled with target architectures.
The ends up with code compiled for `sm_52`, and a just-in-time compile and execution in modern gpus (potentially slower).

In a rudimentary experiment, the preprocessing from centerpoint went down a whole ms, whereas the subsample filter from @manato did not see any reduction.

This can be checked using cuobjdump

### Expected behavior

cuobjdump shows that the code has been compiled for the architectures that autoware is actually used in

### Actual behavior

```text
cuobjdump libautoware_lidar_centerpoint_cuda_lib.so 

Fatbin elf code:
================
arch = sm_52
code version = [1,7]
host = linux
compile_size = 64bit

Fatbin ptx code:
================
arch = sm_52
code version = [8,7]
host = linux
compile_size = 64bit
compressed
ptxasOptions = 

Fatbin elf code:
================
arch = sm_52
code version = [1,7]
host = linux
compile_size = 64bit

Fatbin ptx code:
================
arch = sm_52
code version = [8,7]
host = linux
compile_size = 64bit
compressed
ptxasOptions = 

Fatbin elf code:
================
arch = sm_52
code version = [1,7]
host = linux
compile_size = 64bit

Fatbin ptx code:
================
arch = sm_52
code version = [8,7]
host = linux
compile_size = 64bit
compressed
ptxasOptions = 

Fatbin elf code:
================
arch = sm_52
code version = [1,7]
host = linux
compile_size = 64bit

Fatbin ptx code:
================
arch = sm_52
code version = [8,7]
host = linux
compile_size = 64bit
compressed
ptxasOptions = 
```

### Steps to reproduce

Compile autoware using any version and check with cuobjdump

knzo25, 2025-05-12:
I will create individual PRs for the target packages to avoid reverting the while idea if just one package has issues (should not though)

knzo25, 2025-05-12:
The following PRs address this issue:
 - https://github.com/autowarefoundation/autoware_universe/pull/10611
 - https://github.com/autowarefoundation/autoware_universe/pull/10612
 - https://github.com/autowarefoundation/autoware_universe/pull/10613
 - https://github.com/autowarefoundation/autoware_universe/pull/10614
 - https://github.com/autowarefoundation/autoware_universe/pull/10615

The flags could be defined in the `autoware_package`, but it it the meantime this should be ok (until we know there is no problem for a while).
I added the architectures that autoware is likely/expected to be run in according to:
https://developer.nvidia.com/cuda-gpus

amadeuszsz, 2025-05-14:
@knzo25 
This is great discover!

> I will create individual PRs for the target packages to avoid reverting the while idea if just one package has issues (should not though)

Agree! Let's merge it for now as it is.

What do you think about specifying target architecture dynamically? I checked one package and we can reduce binary size from 15 MB to 13 MB this way. Confirmed architecture with cuobjdump as well.

Could you please check later this PR https://github.com/autowarefoundation/autoware_cmake/pull/29 ?

knzo25, 2025-05-14:
@amadeuszsz 
The problem with any dynamic approach is that you want to distribute a single image right?

amadeuszsz, 2025-05-14:
> [@amadeuszsz](https://github.com/amadeuszsz) The problem with any dynamic approach is that you want to distribute a single image right?

@knzo25 
The current macro has a fallback in which we build for all listed architectures. Anyway, as you pointed correctly, deployed image has to be shipped with all architectures so we can't save here a space with a dynamic approach. We can only optimize it for local builds, but I'm not sure if it has any value - space was never problem over there.