autowarefoundation/autoware_universe #10622: autoware_multi_object_tracker: Mahalanobis Threshold Causes Duplicate Tracking

kaancolak, 2025-05-13:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

Sometimes, especially when tracking stationary buses or trucks, the system creates duplicate tracks for the same object. These duplicate tracks tend to appear and disappear suddenly.

In the video below, I logged and displayed Mahalanobis distance values observed when a duplicate issue occurs.

Currently, the default Mahalanobis threshold value is set to 3.035, which corresponds to the 99th percentile of the chi-squared distribution with 2 degrees of freedom. This setting is intended to filter out moderate outliers.

🔗 [Code Reference – Line 183](https://github.com/autowarefoundation/autoware_universe/blob/main/perception/autoware_multi_object_tracker/lib/association/association.cpp#L183)

However, real-world data shows that the Mahalanobis distance for detections of the same object can reach up to ~3.51

Check the 8th second for duplicate tracking:

https://github.com/user-attachments/assets/95f8e311-8de8-4413-83cf-a4f70f1d49ba

Detection results for the same scene: 

https://github.com/user-attachments/assets/32e3fb34-92a0-49c8-ab8d-50c7cfe9010e

TrackerProcessor::removeOverlappedTracker does not work as expected because duplicated track objects have recent updates.

🔗 [Code Reference – Line 186](https://github.com/autowarefoundation/autoware_universe/blob/main/perception/autoware_multi_object_tracker/src/processor/processor.cpp#L186)

*Possible Solutions:*

*Option 1:*  
To better accommodate real-world variation, we can increase the Mahalanobis threshold to ~3.461, which corresponds to the 99.7th percentile. This setting is intended to filter out strong outliers.

*Option 2:*
Update the removeOverlappedTracker function logic to better handle the removal of highly overlapped objects

FYI: @YoshiRi @technolojin @miursh @yukkysaito 

### Expected behavior

Tracked objects continuously without duplication.

### Actual behavior

The autoware_multi_object_tracker creates duplicate tracks for the same object.

### Steps to reproduce

Run Autoware.

technolojin, 2025-05-14:
@kaancolak 
Thank you for your analysis. @ogrex and I also see similar situation.

@ogrex 
I would like to request to share your analysis result, and suggest appropriate threshold value.

ogrex, 2025-05-22:
### Analysis and Tuning for Yaw Stability based on Mahalanobis Threshold together with Q,R

We investigated the impact of yaw-related parameter tuning on the consistency and robustness of EKF-based state estimation, especially under sharp vehicle turns. The evaluation focused on three primary strategies:

---

#### (1) Tuning Measurement Covariance for Yaw (`R`)

By increasing the measurement covariance for yaw, we reduce the EKF’s sensitivity to rapid yaw changes, which helps prevent overestimating the error during real-world turning maneuvers.

- **Observation**: With the default covariance of `20`, some Mahalanobis distances exceeded the gating threshold.
- **Adjustment**: Setting `R.yaw = 40` generally brings most cases within threshold.
- **Caution**: An excessively large value may result in under-correction, so it's important to strike a balance between robustness and measurement responsiveness.

---

#### (2) Adjusting Process Noise for Yaw Rate (`Q`)

Increasing process noise allows the EKF to expect higher variability in yaw rate during the prediction phase. This results in a more forgiving prediction covariance, reducing the relative innovation (residual) when compared to actual sensor data.

- **Trade-off**: While higher process noise improves turn performance, it may degrade accuracy during straight-line driving. Proper tuning is critical to maintain overall estimation quality.

---

#### (3) Revisiting the Gating Threshold

We experimented with increasing the Mahalanobis gating threshold from the default to around **3.4–3.5**.

- **Outcome**: This adjustment improved acceptance of measurements overall, though a few edge cases still exceeded the gate.
- **Risk**: Loosening the gate increases the chance of incorporating outliers, so this must be carefully managed.

---

#### (4) (Future Consideration) Dynamic Parameter Tuning

For even better adaptability, we may consider **temporarily relaxing parameters like yaw covariance or gating thresholds** when sharp turns are detected (e.g., based on high yaw rate). This could be implemented via dynamic reconfiguration or model switching to maintain precision during straight motion and flexibility during turns.

---

### Evaluation Focus

For this round of evaluation, we focused on **(1)**, **(2)**, and **(3)**.

---

### Experiment Results

We compared different settings of `R.yaw` (measurement covariance) and `Q.yaw_rate_max` (process noise) by evaluating the **maximum Mahalanobis distance** across several scenarios:

| R.yaw | Q.yaw_rate_max | Max Mahalanobis Distance (samples)                  |
|-------|----------------|------------------------------------------------------|
| 40    | 15             | 2.24, 2.68, 2.28, 3.12, 3.30, 3.35                   |
| 20    | 15             | 2.28, 3.20, 2.58, 3.28, 3.62, 3.74 *(default)*       |
| 5     | 15             | 2.80, 5.67, 4.59, 4.14, 6.75, 6.27 *(too strict)*    |
| 20    | 30             | 1.84, 2.51, 2.21, 3.03, 3.34, 2.96 *(effective)*     |
| 20    | 20             | 1.94, 2.65, 2.24, 3.18, 3.32, 3.10                   |

---
### Impact of Gate Threshold Setting

To analyze the effect of different gate thresholds, we plotted the current measurement points alongside the current `Q` and `R` configuration in a diagram.  

In this plot, any point that lies **above and to the right of the QR setting line** indicates a case where the Mahalanobis distance **exceeds the threshold**.

This visualization helped guide the appropriate adjustment of the gating threshold, ensuring that parameter tuning remains effective while avoiding unnecessary rejections.
Tuning (1)R, (2)Q, (3)Threshold
3.035 (Default)
[image]
3.4
[image]
3.6
[image]

---

### Tuning Strategy: Gradient-Based Adjustment

To minimize system perturbation while improving gating performance, we applied a **gradient-based tuning approach**. Specifically:

- The Mahalanobis distance gate threshold was temporarily increased to **3.42**.
- Parameters `R` (measurement covariance) and `Q` (process noise) were then fine-tuned **along the gradient direction**, aiming to balance sensitivity and robustness.
- This method helped suppress abrupt parameter-induced fluctuations while converging on a more stable and effective configuration.

📌 See related commit:  
[`feat(multi_object_tracker): parameters tuned for mahalanobis distance`](https://github.com/autowarefoundation/autoware_universe/commit/020d03d)

### Conclusion

From the experiment, we observed that by carefully adjusting `Q` (process noise) and `R` (measurement covariance), the Mahalanobis distance can be tuned to stay around **3.4**, which helps minimize disruptive changes to the Mahalanobis threshold.

However, since modifying `Q` and `R` can impact overall tracking behavior, a threshold of **3.7** appears to be a temporarily appropriate and balanced choice — aligning well with the analysis results while ensuring system stability.

closed 2025-05-27 by technolojin