autowarefoundation/autoware_universe #10932: Make Autoware Universe CI/CD faster

mitsudome-r, 2025-07-07:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

Currently, autoware_universe requires build-and-test-differential to pass before merging, but it sometimes takes a long time to wait for the CI to pass, especially when you make modification to packages that depends on many other packages.

I would like to think of a solution to make the CI/CD to run faster by utilizing build cache of dependent repositories.

### Purpose

Make build-and-test-differential CI faster. 

### Possible approaches

1. Create a workflow to create a underlying colcon ws (dependency_ws) for build_depends.repos and push the cache to GitHub cache.This workflow can run daily or can be triggered manually.
2. Modify the build-and-test-differential workflow to source the underlying workspace before colcon build and colcon tests.

### Definition of done

1. build_dependent_workspace workflow is merged
2. modify build-and-test-differential workflow to use the dependency_ws

xmfcx, 2025-08-05:
- https://github.com/autowarefoundation/autoware_universe/blob/main/build_depends_nightly.repos
- https://github.com/autowarefoundation/autoware_universe/blob/main/build_depends_humble.repos

- Build cache: https://github.com/autowarefoundation/autoware_universe/actions/caches

For build depends repos, we can cache build and install folders from these repositories and load

xmfcx, 2025-08-06:
Yes, primarily that's the goal. The tricky part is to decide when to rebuild them.
- a new workflow vs adding it to an existing workflow
- how to detect changes in target branches/heads of repositories
  - event based or how often to check if periodical

esteve, 2025-08-06:
@xmfcx I'm going to start by creating a new workflow to avoid interfering with the existing ones, but if later on we see that it's better to integrate it into another one, we can do that too. Thanks for the explanation.

xmfcx, 2025-08-15:
On a separate note, I've checked that the cache generated from workflow_dispatch on main can be accessed from the PRs with this dummy repo:
- [this workflow creates cache](https://github.com/xmfcx/cache-test/blob/1b9c3fa962ab9790bf9dfb8cd46fb71601d32c31/.github/workflows/seed-cache.yml)
- [this workflow uses the cache](https://github.com/xmfcx/cache-test/blob/1b9c3fa962ab9790bf9dfb8cd46fb71601d32c31/.github/workflows/use-cache.yml)
- [it is restored here successfully ✅](https://github.com/xmfcx/cache-test/actions/runs/16997001743/job/48190007394?pr=1#step:3:19)