autowarefoundation/autoware_universe #10963: Support Bus Stop Behaviors

mehmetdogru, 2025-07-10:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

This issue created to host and track bus stop related PRs

PRs:
- https://github.com/autowarefoundation/autoware_lanelet2_extension/pull/71

---

PRs will specifically follows the design as follows:

[image]

### Purpose

Support some features like pull-over and pull-out behavior for bus stops that has custom centerline  

### Possible approaches

Modify `start_planner` and `goal_planner`  packages to support bus stops

### Definition of done

- [ ] Add pull-over signalling for bus stops
- [ ] Add pull-out behavior (e.g. signalling & safety check) for bus stops

soblin, 2025-07-11:
At the first glance, I thought we can just substitute intersection lanelet (the lanelet with `turn_direction` tag) for achieving behavior_planning/motion_planning, and the proposed `bus_stop` tag for the lanelet is used only for manually setting the goal in Psim on Rviz GUI during development. In operation we will use routing service, in which case route segment is directly set to mission_planner. Can you elaborate why we need this tag in service operation ?

mehmetdogru, 2025-07-11:
> At the first glance, I thought we can just substitute intersection lanelet (the lanelet with `turn_direction` tag) for achieving behavior_planning/motion_planning, and the proposed `bus_stop` tag [...]

Thank you @soblin for your comment.

IMO, there are couple of reasons I can think of why we would need `bus_stop` tag and why we need to use `start_planner` instead of `intersection_module. So please let me explain why I think so:

1. The use case is much more similar to `start_planner` than `intersection` meaning we need to consider cars approaching behind like in `start_planner` before we pull out
2. We are not technically in an intersection, we are just pulling out so again semantically `start_planner` as intended is more appropriate 
3. Intersection module does not perform well on overlapping lanelets as it can be the case for bus stop most of the time. (As I tried to explain with the picture in description why we need overlapping lanelets is mainly to be able to use custom centerlines for bus stop lanelets)
4. I will be creating more PRs today, sorry they are not in the issue yet but there are some specific requirements for signalling and starting the drive related to this task as well so it is quite challenging and less tidy to achieve those tasks with using intersection module

mehmetdogru, 2025-07-11:
> Thank you for the proposal nad changes 😍
> 
> Is the motivation for wanting to run start_planner in the bus stop area because currently start_planner doesn't activate when departing from a bus stop that is on a road?
> 
> Please let me confirm whether related changes are to handle cases where regulatory element (subtype: bus_stop_area) references lanelet's subtype is road not road shoulder.

Thank you for your comment @kyoichi-sugahara 😊 
Please forgive me if I understood your question wrong and please do let me know if my reply does not answer your question 🙇 

Technically `bus_stop_area` reg element is used for `goal_planner` to be able to contain candidate goal poses rather than `start_planner`.  However `bus_stop` lanelet tag serves the purpose of defining bus stop lanelets so that pull out safety check and signalling behavior can be adjusted. `start_planner` does not run if ego is on centerline but if ego is on `bus_stop` lanelet, we need to perform yield if necessary (safety check) and signal when we pull out even though ego is on a centerline.

kyoichi-sugahara, 2025-07-11:
@mehmetdogru 

You basically understand it correctly. Thank you for understanding this even though the documentation isn’t organized well.

You’re right that if the ego vehicle is at the center of the road, start_planner won’t start, but if that lanelet is defined as road_shoulder, start_planner will activate.

However, I think it’s common to want it to activate when there’s a bus stop even if it’s not road_shoulder, so from a start_planner maintainer’s perspective, I’m fully in favor of this proposal.​​​​​​​​​​​​​​​​