autowarefoundation/autoware_universe #11418: Autoware.universe support ROS2.JAZZY

liuXinGangChina, 2025-09-29:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

We will make some necessary modification to current autoware code to make it work with ros2 jazzy.
In order to achieve that, we should follow below steps:

    add ci support for ros jazzy
    edit current core code to make it pass ci check
    support humble and jazzy at the same tim

### Purpose

make autoware.universe work with ros2 jazzy.

### Possible approaches

1.     add ci support for ros jazzy(add docker build process and pr check process)
2.     edit current core code to make it pass ci check

### Definition of done

- [ ] 1. add ci support for ros jazzy
- [ ] 2. edit current core code to make it pass ci check

xmfcx, 2026-01-16:
**Waiting for review:**
- https://github.com/autowarefoundation/autoware_universe/pull/11872
- https://github.com/autowarefoundation/autoware_universe/pull/11874
- https://github.com/autowarefoundation/autoware_universe/pull/11878
- https://github.com/autowarefoundation/autoware_universe/pull/11880
- https://github.com/autowarefoundation/autoware_universe/pull/11882
- https://github.com/autowarefoundation/autoware_universe/pull/11883

Once these are merged, status in autoware_universe:

[image]

xmfcx, 2026-01-20:
- https://github.com/autowarefoundation/autoware_universe/pull/11924
- https://github.com/autowarefoundation/autoware_universe/pull/11925
- https://github.com/autowarefoundation/autoware_universe/pull/11926
- https://github.com/autowarefoundation/autoware_universe/pull/11927
- https://github.com/autowarefoundation/autoware_universe/pull/11928

Once these are merged, status in autoware_universe:

[image]

liuXinGangChina, 2026-02-10:
https://github.com/autowarefoundation/autoware/actions/runs/21821227429/job/62954780440#step:7:21767
Now universe-jazzy series docker build process failed, it is because autoware use universe-0.49.0 while jazzy support prs recentlly were on main branch of universe. Once universe version bumped, docker build process will succeed.

Currently fails on control/smart_mpc relies on python3-torch while jazzy dose not have yet, so conditional install by apt is set in main branch of universe

[image]

[image]

on universe 0.49.0 

[image]
 while on universe main

[image]